hjam40/Camera.MAUI #168: version 1.5.1 does not work on ios

maslewandowski, 2024-07-10:
The event for qr code recognition is not fired when holding a qr code from a business card right in front of the camera.

see [this repository](https://github.com/maslewandowski/cameramauiproblem) to get the code.
i have invested a bit time trying to resolve the problem but i dont get it working.

interesting is, that with the version 1.2.1 the code is working fine and the event gets fired.

am i doing something wrong or is there a bug in this version?

thanks!

connorjarred, 2024-07-10:
I also discovered this yesterday that the Camera.Maui versions 1.5.0 and 1.5.1 barcode reader are also not detecting the barcode event trigger for EAN13 codes on Android.
All the Camera.Maui  versions from 1.2.1 to 1.4.4 are working fine.
The versions of the package Camera.Maui up to 1.4.4 are dependent on the ZXing.Net package (0.16.9) and the Camera.Maui  package versions above this from 1.5.0 do not reference the ZXing.Net package anymore, so I think that the external ZXing.Net package is now built in into the Camera.Maui package or its using another software mechanism.
This is the description of the ZXing.Net package from nuget:
ZXing.Net is a port of ZXing, an open-source, multi-format 1D/2D barcode image processing library originally implemented in Java.
      It has been ported by hand with a lot of optimizations and improvements.
      It is compatible with .Net 2.0, .Net 3.5, .Net 4.x, .Net 5.x, .Net 6.x, .Net 7.x, Windows RT Class Library and Components, UWP, .Net Standard 1.x and 2.0x, .Net Core App 3.x, Silverlight 4, Silverlight 5, Windows Phone 7.x and Windows Phone 8.x and Xamarin.Android

Hope this info might help whoever has time to look into the source code.
Greetz,
Connor Jarred

maslewandowski, 2024-07-10:
> All the Camera.Maui versions from 1.2.1 to 1.4.4 are working fine.

I will check this out, thanks.

connorjarred, 2024-07-10:
@maslewandowski I solved it for Android.
I use Camera.MAUI nuget package 1.5.1 and added the Camera.MAUI.ZXing nuget package (1.0.0 only verssion) to my solution (VS 2022).
Then in my simple MainPage.xaml.cs added the usings and default code. I'm using the BarcodeFormat.EAN_13 but the Qcode option also worked 4 me.
I've added the  VS solution as a Zip file to this comment.
[TestBarcode.zip](https://github.com/user-attachments/files/16166895/TestBarcode.zip)

Hope this helps you with your iOS problem as well.

Good luck!