hjam40/Camera.MAUI #170: `NullReferenceException` inside `cameraView.TakePhotoAsync`

ITaluone, 2024-07-17:
Hi

We an issue with this to take a photo, because it is throwing an NRE somewhere inside this method, but I cannot tell where exactly.

Platform: Android
Framework: .NET8.0

This is our code in question:

```csharp
private async Task<Stream> TakePhotoAsync()
{
    // ... pre-checks 

    var photoResult = await cameraView.TakePhotoAsync(Camera.MAUI.ImageFormat.PNG);

    if (photoResult == null)
    {
        Debug.WriteLine("Error: photoResult is null.");
        return null;
    }

    var stream = new MemoryStream();
    await photoResult.CopyToAsync(stream);
    stream.Position = 0;
    return stream;
}
```

I cannot give more information, unfortunately, because of lack of stack trace and such.

Can you give me a hint what this causes?

luresti, 2024-07-18:
i suggest you select  the camera using next code to assure to use the correct position camera 

for (int i = 0; i < cameraView.Cameras.Count; i++)
{
    if (cameraView.Cameras[i].Position == Camera.MAUI.CameraPosition.Front)
    {

        cameraView.Camera = cameraView.Cameras[i];
        break;
    }

}

ITaluone, 2024-07-19:
Ahm.. I only have two cameras back and front. I want to select the back camera (which is the first one) but when I select it, it crashes.

luresti, 2024-07-24:
change CameraPosition.Front    to CameraPosition.Back