hjam40/Camera.MAUI #174: Memory leaking

douglassimaodev, 2024-08-20:
I am just referring the camera in my page

 <cv:CameraView x:Name="bcReader" 
                HorizontalOptions="Start"
                HeightRequest="350" />
                
  that is alll, no extra element and no event binding to it, so when I navigate from page A to page B (where the CamareView is) when I go back to page A CameraView is still alive, and I didnt even start the camera, as I said, there is just an element in the page but nothing in the code behind to keel it alive

DanKyungu, 2024-08-22:
Did you tried to Disconnect the CameraView Handler on the page Unloaded event?

alexthecoolman, 2025-03-28:
Same for me, memory leak spoted by https://github.com/AdamEssenmacher/MemoryToolkit.Maui.