hjam40/Camera.MAUI #176: Camera not loading on some Samsung devices

tomovski, 2024-09-03:
After updating to .net 8.0.3 version the camera stopped loading for some Samsung devices. And it's only happening on the first camera 
`cameraView.Camera = cameraView.Cameras.First();`
While it is working on other Samsung devices, Pixel and iPhone. 

I also receive a success result in the StartCameraAsync, but nothing is rendered.
`var result = await cameraView.StartCameraAsync();`

I've also tried the the workaround with:
```
await cameraView.StopCameraAsync();
var result = await cameraView.StartCameraAsync();
```
and tried with some small delay before stopping/starting the camera, but nothing seems to work. 

I am using the latest version of Camera.MAUI and Camera.MAUI.ZXing. I saw some other people also facing the issue.
Is there any other workaround ? Many thanks

jefterdorea, 2024-09-23:
I'm also experiencing this problem.

Interestingly, the problem has already occurred with Samsung and Motorola devices. But for the vast majority of users, everything is working perfectly.

No error is displayed, it just doesn't display the screenshot.

moze-max, 2024-10-07:
你可以试试先运行`await cameraView.StopCameraAsync();`，然后再运行`await cameraView.StartCameraAsync();`

jefterdorea, 2024-10-09:
Thank you very much @moze-max , it worked very well. Simple as that.