hjam40/Camera.MAUI #178: StopCameraAsync does not remove green dot and turn off the light front of the camera on IOS

hercul1017, 2024-09-19:
Hi,

After moving to another pages the green dot indicating the camera still on does not disappear even after calling StopCameraAsync. I tried following code but it did not work. StopCameraAsync returns Success, but green dot continues to display. Newer models of IPhones light next the front camera that I use, also continues to light and it does not turn off.

        MainThread.BeginInvokeOnMainThread(async () =>
        {
            cameraView.FlashMode = FlashMode.Disabled;
            cameraView.TorchEnabled = false;

            if (await cameraView.StopCameraAsync() == CameraResult.Success)
            {                 
            }            
        });
      
Thanks for any help,

thomasgalliker, 2024-10-08:
I guess this comes from the fact, that MAUI does not call DisconnectHandler automatically when a page (and all its UI elements) is no longer used. I created this helper class which automatically disconnects handlers once the page is navigated away: [HandlerCleanUpHelper.cs](https://github.com/thomasgalliker/CameraScanner.Maui/blob/develop/CameraScanner.Maui/Utils/HandlerCleanUpHelper.cs)

Someone who maintains Camera.MAUI would have to take something like that into consideration. MAUI will probably not solve the problem for us...

Ducvu1996, 2025-04-11:
protected override async void OnDisappearing()
     {
         base.OnDisappearing();
         try
         {
             await SCanQRCode.StopCameraAsync();
         }
         catch (Exception)
         {
         }
     }
try to use it