hjam40/Camera.MAUI #189: cv:CameraView.Cameras.Count = 0 on Release, but Debug is can use.

Gao996, 2024-12-05:
My AndroidManifest.xml:
<?xml version="1.0" encoding="utf-8"?>
<manifest xmlns:android="http://schemas.android.com/apk/res/android" android:versionName="1.0" android:versionCode="1">
	<application android:allowBackup="true" android:icon="@mipmap/appicon" android:supportsRtl="true" android:label="上课系统"></application>
	<uses-permission android:name="android.permission.ACCESS_NETWORK_STATE" />
	<uses-permission android:name="android.permission.INTERNET" />
	<uses-permission android:name="android.permission.CAMERA" />
	<uses-permission android:name="android.permission.RECORD_AUDIO" />
	<uses-permission android:name="android.permission.RECORD_VIDEO" />
	<uses-sdk android:minSdkVersion="29" android:targetSdkVersion="29" />
</manifest>
Debug is can use, Release no use.
Camera.Maui is 1.2.1 version.
.net core 7.0.

ReneKraus, 2024-12-20:
I have the same problem with my MAUI app for Windows. I am using version 1.5.1 with .NET 8. I have already disabled code optimizations and set the DeviceCapability for webcam and microphone.

Blue-RiverZJ, 2025-01-05:
Release+.net8 also cannot be used. It crashes after running.