hjam40/Camera.MAUI #190: Duplicate Entry

FANMixco, 2025-01-02:
I'm wondering if I'm using wrongly the library since I'm getting this error:

> 1>...\ai-chef-maui\MainPage.xaml(25,28): XamlC warning XC0022: Binding could be compiled to improve runtime performance if x:DataType is specified. See https://learn.microsoft.com/dotnet/maui/fundamentals/data-binding/compiled-bindings for more information.
> 1>ai-chef-maui -> ...\ai-chef-maui\bin\Debug\net9.0-windows10.0.19041.0\win10-x64\ai-chef-maui.dll
> 1>WINAPPSDKGENERATEPROJECTPRIFILE : error : PRI175: 0x80073b0f - Processing Resources failed with error: Duplicate Entry.
> 1>WINAPPSDKGENERATEPROJECTPRIFILE : error : PRI277: 0x80073b0f - Conflicting values for resource 'Files/Microsoft.Maui.Controls/Compatibility/Handlers/TableView/Windows/TableViewStyles.xbf'

This is a new app, I haven't added any extra NuGets or anything.

[image]

Thanks.

jasells, 2025-01-02:
@FANMixco   I'm just starting to look into some other issues in this package as well, bit I see you're building your conuming app in .Net9 (I think from your log output)? `...Debug\net9.0-windows10.0.19041.0\win10-x64...`

This repo looks like it started as a .Net7 only lib, then added .Net8.  Maybe try retargeting you app project to .Net8 just for grins?  There may be something that needs updated in this lib to support .Net9?

FANMixco, 2025-01-02:
> @FANMixco I'm just starting to look into some other issues in this package as well, but I see you're building your consuming app in .Net9 (I think from your log output)? `...Debug\net9.0-windows10.0.19041.0\win10-x64...`
> 
> This repo looks like it started as a .Net7 only lib, then added .Net8. Maybe try retargeting your app project to .Net8 just for grins? There may be something that needs updated in this lib to support .Net9?
> 
> I have successfully built a Maui 8 app (debug at least) with this package for Win/iOS/Android.

Indeed, it was something about .NET 9. Thanks! I recreated it for .NET 8.

jasells, 2025-01-02:
> Indeed, it was something about .NET 9. Thanks! I recreated it for .NET 8.

Nice! 

I don't think .Net9 supports building .Net7 targeted projects anymore, due to MS's support [strategy/policy](https://dotnet.microsoft.com/en-us/platform/support/policy/dotnet-core).  I am pretty sure I've seen similar build issues in some of my pipelines that still have .Net7 targets when I tried to use "latest" SDK (.Net 9) instead of "Latest .Net 8 SDK." I _think_ early preview releases of .Net9 would still build .Net7 projects, but it was dropped sometime this summer before the official .Net9 release.

Guessing windows target in this [lib](https://github.com/hjam40/Camera.MAUI/blob/bc9dcb08009633a2fdff37d75494374d988a3369/Camera.MAUI/Camera.MAUI.csproj#L5): `net7.0-windows10.0.19041.0` is the issue and needs to be dropped, and possibly update the windows version too to make .Net9 happy?

This is the minimum (Windows 10 [version 1809](https://github.com/MicrosoftDocs/windows-dev-docs/blob/docs/hub/apps/winui/winui3/release-notes/winui3-project-reunion-0.5.md)) [specified](https://learn.microsoft.com/en-us/dotnet/maui/supported-platforms?view=net-maui-9.0) by Maui/.Net 9, so I don't know without poking at it some.  You would _think_ that the build system would just ignore the .Net 7 libs in the package...
```
<TargetPlatformMinVersion Condition="$([MSBuild]::GetTargetPlatformIdentifier('$(TargetFramework)')) == 'windows'">10.0.17763.0</TargetPlatformMinVersion>
```

FANMixco, 2025-01-03:
> Nice!
>
> I don't think .Net9 supports building .Net7 targeted projects anymore, due to MS's support [strategy/policy](https://dotnet.microsoft.com/en-us/platform/support/policy/dotnet-core). I am [...]

I can work with .NET 8 for this project. Thanks for the suggestion. However, I can help you test with .NET 9.

jfversluis, 2025-01-09:
Can you try including a reference to Microsoft.Maui.Controls.Compatibility?

JohnnySWG, 2025-02-03:
May I ask an update on this? 

I have encountered the same build problem with a .NET Maui app that uses to .NET 9.

Will there be support for .NET 9?

Anawaz, 2025-04-08:
I am also facing same issue with .net9 project. 
`PRI175: 0x80073b0f - Processing Resources failed with error: Duplicate Entry.
PRI277: 0x80073b0f - Conflicting values for resource 'Files/Microsoft.Maui.Controls/Platform/Windows/CollectionView/ItemsViewStyles.xbf'`

DamianMehers, 2025-05-07:
> Microsoft.Maui.Controls.Compatibility

This resolved the error for me.

jasells, 2025-10-03:
There are forks of this project that have been updated, and fix other issues as well.
https://github.com/jasells/CameraMaui
https://github.com/janusw/CameraMaui

ZarehD, 2025-10-03:
Thank you, @jasells. That's very helpful. I'll check them out.