hjam40/Camera.MAUI #193: Is there a way to focus a pic in Android?

FANMixco, 2025-01-31:
I have added the plugin perfectly and works nice. However, I noticed that when I run it from my phone, it seems the Viewer cannot focus what it's "viewing":

[image]

This is how it looks in the normal camera:

[image]

Is there any option that I should enable? I know that in the phones, you can often tap the viewer to refocus, but I'm not sure if it's possible. Thanks.

Philigane, 2025-03-04:
Stop it before starting it even if you didn't start it yet

IljaKan, 2025-08-15:
I am experiencing the same issue on Samsung Ultra S25, picking the first camera.

IljaKan, 2025-08-15:
> I am experiencing the same issue on Samsung Ultra S25, picking the first camera.

On ios it focuses perfectly