hjam40/Camera.MAUI #199: **System.ObjectDisposedException:** 'Cannot access a disposed object. Object name: 'Camera.MAUI.Platforms.Android.MauiCameraView'.'

rahulshreebastav, 2025-05-14:
[ViewRootImpl@850ba90[MainActivity]]   sfl=100000}
[ViewRootImpl@850ba90[MainActivity]] updateBlastSurfaceIfNeeded mBlastBufferQueue=0x749f494600 isSameSurfaceControl=true
[BLASTBufferQueue] update, w= 1080 h= 2400 mName = ViewRootImpl@850ba90[MainActivity] mNativeObject= 0x749f494600 sc.mNativeObject= 0x749fa978e0 format= -1 caller= android.view.ViewRootImpl.updateBlastSurfaceIfNeeded:2898 android.view.ViewRootImpl.relayoutWindow:9847 android.view.ViewRootImpl.performTraversals:3884 android.view.ViewRootImpl.doTraversal:3116 android.view.ViewRootImpl$TraversalRunnable.run:10885 android.view.Choreographer$CallbackRecord.run:1301 
[ViewRootImpl@850ba90[MainActivity]] Relayout returned: old=(0,0,1080,2400) new=(0,0,1080,2400) req=(1080,2400)0 dur=16 res=0x0 s={true 0x74b2737800} ch=false seqId=0
[ViewRootImpl@850ba90[MainActivity]] ViewPostIme key 0
[Activity] onKeyDown(KEYCODE_BACK)
[ViewRootImpl@850ba90[MainActivity]] ViewPostIme key 1
[Activity] onKeyUp(KEYCODE_BACK) isTracking()=true isCanceled()=false hasCallback=false
[ltech.traze_app] Explicit concurrent copying GC freed 353KB AllocSpace bytes, 2(104KB) LOS objects, 49% free, 8872KB/17MB, paused 181us,130us total 88.999ms
[OnBackInvokedCallback] OnBackInvokedCallback is not enabled for the application.
[OnBackInvokedCallback] Set 'android:enableOnBackInvokedCallback="true"' in the application manifest.

**System.ObjectDisposedException:** 'Cannot access a disposed object.
Object name: 'Camera.MAUI.Platforms.Android.MauiCameraView'.'

After updating MAUI controls to 9.0.70, I get these exceptions.

DeanZhuo, 2025-07-08:
Here with the same issue after updating to net9

rahulshreebastav, 2025-07-08:
Hi,

Please update the Camera.MAUI package to the latest version (1.5.1). Additionally, before starting the camera, add the following line of code:

cameraView.BarCodeDecoder = new ZXingBarcodeDecoder();
This class is already available in the referenced project. The final setup should look like this:

cameraView.Camera = cameraView.Cameras.First();
cameraView.BarCodeDecoder = new ZXingBarcodeDecoder();
MainThread.BeginInvokeOnMainThread(async () =>
{
    await cameraView.StopCameraAsync();
    await cameraView.StartCameraAsync();
});